SJSURoboticsTeam/urc-intelligent-systems-2023 #17: Obstacle Avoidance

acarrou, 2022-11-14:
The rover needs some way to avoid obstacles when we run the central autonomous drive system. This will involve different sensors such as lidar, cameras, and potentially more. We want to stay away from ROS and stick with python as this is what we have been working with for autonomous driving. The idea is to re-path the rover to the GPS coordinate when there is an obstruction.

acarrou, 2023-04-22:
Lidar is being worked on with #53 and #54 but still needs to implement point cloud with the camera.

acarrou, 2023-05-10:
@ElaynaSeguin @dontario Main needs to be tested for object detection. Use the ObjectDetector.py class and make a demo.

chrehall68, 2024-03-30:
We have pivoted to something similar to this with the newest revision of the repository.

closed 2024-03-30 by chrehall68