SJSURoboticsTeam/urc-intelligent-systems-2023 #72: Mock Sensor Implementation

Maaz0070, 2023-07-26:
## GitHub Issue: Create Mock Sensors for Autonomous Systems Project

**Description:**
As part of enhancing the testing capabilities of the Autonomous Systems project, we need to create mock sensor classes for all the sensors used in the project. These mock sensor classes will simulate the behavior of real sensors by providing random data for testing various functionalities.

**Mock Sensor Implementation:**

1. **MockBN08x:**
   - Simulates an IMU sensor (BN08x) and provides random acceleration, gyro, magnetic, and rotation values.

2. **MOCKLSM303:**
   - Simulates an LSM303 sensor and provides random acceleration and magnetic field values.

3. **MOCKWiFi:**
   - Simulates a WiFi module and allows the testing of WiFi-related functionalities.

4. **MOCKGPS:**
   - Simulates a GPS sensor and provides random latitude, longitude, altitude, and velocity values.

5. **MOCKMPU6050:**
   - Simulates an MPU6050 sensor and provides random acceleration and gyro values.

6. **MOCKRPLIDAR:**
   - Simulates an RPLIDAR sensor and allows the testing of LiDAR-related functionalities.

7. **MOCKSerial:**
   - Simulates a serial communication module and provides random data transmission behavior.

8. **MOCKIR_Sensor:**
   - Simulates an IR sensor and provides random distance/proximity values.

9. **MOCKStereoCamera:**
   - Simulates a stereo camera and provides random image data for testing computer vision functionalities.

**Steps:**

1. Create the mock sensor classes in the `modules` directory of the project.

2. Implement the necessary methods in each mock sensor class to provide random data similar to the real sensors.

3. Create a separate branch named `feature/mock-sensors` branching off from the `main` branch.

4. Commit the new mock sensor classes to the `feature/mock-sensors` branch.

5. Test the mock sensors by writing unit tests in the respective test files.

6. Once the mock sensors are fully tested and verified, create a pull request to merge the `feature/mock-sensors` branch into the `main` branch.

**File Changes:**
Autonomous_Systems/
└── modules/
├── BN08x.py
├── GPS.py
├── IR_Sensor.py
├── LSM303.py
├── MOCKBN08x.py # New
├── MOCKLSM303.py # New
├── MOCKWiFi.py # New
├── MOCKGPS.py # New
├── MOCKMPU6050.py # New
├── MOCKRPLIDAR.py # New
├── MOCKSerial.py # New
├── MOCKIR_Sensor.py # New
├── MOCKStereoCamera.py # New
├── MPU6050.py
├── RPLIDAR.py
├── Serial.py
├── WiFi.py

**Testing:**
- Write comprehensive unit tests for each mock sensor class to ensure they provide accurate and random data as expected.

**Expected Outcome:**
- The mock sensor classes should be able to mimic the behavior of the real sensors, providing random data for testing different functionalities in the Autonomous Systems project.

**Additional Notes:**
- This enhancement will significantly improve the project's testability and enable better validation of various functionalities.
- Once the pull request is approved and merged, we can confidently conduct testing and development in isolated environments using the mock sensors.
- Feel free to provide any suggestions or feedback on the mock sensor implementation during the review process.
- Please let us know if you have any questions or concerns regarding this task.

**Future Scope**
- Look into making a mock scenario file that mimics the rover moving closer to a target GPS position

LucasStevenson, 2023-08-06:
i ran the mock classes through the test files in `demos/` and they all worked fine

closed 2023-08-06 by LucasStevenson

suhailskhan, 2023-08-22:
> **Future Scope**
- Look into making a mock scenario file that mimics the rover moving closer to a target GPS position

Possible approach, as discussed with @Maaz0070 :
Record a sample of actual sensor values while rover is moving to a GPS coordinate.